imeszaros/ledclock #1: Issues flashing firmware on ESP8266

Bubliok, 2022-05-22:
I am unable to run the firmware on my ESP8266. It seems to flash normally but the AP is not working (also no status led). I am unable to connect and setup the module. I tried the pre-compiled release and also tried compiling it myself using d1_mini environment but no luck with that either. The original WLED firmware works fine and I am able to connect to the AP.

imeszaros, 2022-05-23:
Hey @Bubliok , I targeted ESP32 with my firmware because I had some problems on the 8266. I cant remember exactly that the trouble was operating the piezo speaker simultaneously either with the LEDs or the ambient light sensor, because the 8266 had not enough timers maybe?

I'll have a look on it and maybe create a 8266 env and disable the piezo speaker probably.

imeszaros, 2022-05-23:
Please checkout the branch `esp8266` : https://github.com/imeszaros/ledclock/tree/esp8266

Switch to the `ledclock_esp8266` environment and build the project, and upload it to your ESP8266 board. I was able to upload and connect to it, but had not enough time to sketch the circuit on a breadboard. Speaker is disabled as of now.

If it works for you I may try to get the speaker working or remove the relevant UI sections if it can't be done.

Bubliok, 2022-05-23:
It works now! I was able to compile it and upload it. Thanks!

closed 2022-09-25 by imeszaros