imeszaros/ledclock #2: Can you share the wiring of diagram of the connections?

razserv2010, 2022-08-03:
Can you take a picture of the soldering and arrangement of the leds?
I printed it but I don't know how to arrange the LEDs with the solders (I can't find an exact explanation on github)
And do segments need to be defined?

imeszaros, 2022-08-12:
Sorry for the late reply! I attach an image of the led arrangement.

[image]

You do not need to define segments by default. I hope this helps, let me know if you need more info!

razserv2010, 2022-08-12:
Thanks, I think it's fine
Where did you connect the data of the LED and the buzzer? in esp32

imeszaros, 2022-08-12:
You can find the schematic and PCB here: https://github.com/imeszaros/ledclock/tree/master/ledclock/sch-pcb

Wiring of the ESP can be found here: https://github.com/imeszaros/ledclock/blob/master/ledclock/sch-pcb/schematic-main-board.pdf

imeszaros, 2022-08-16:
@razserv2010 did you managge to build the clock? If yes, would you mind sharing some pictures? I'd be happy to see it!

razserv2010, 2022-08-16:
> @razserv2010 did you managge to build the clock? If yes, would you mind sharing some pictures? I'd be happy to see it!

I haven't managed to build it yet, I hope that this weekend I will have time, I will update with pictures the installation process..

closed 2022-09-25 by imeszaros