imeszaros/ledclock #5: How can I change the number of LEDs / order of LEDs?

imeszaros, 2022-10-03:
**If you want to change how many LEDs make up a segment:**

- change the total number of LEDs in [platformio_override.ini](https://github.com/imeszaros/ledclock/blob/e31a81c1ca2a10eb4417e8a380c6760602ac45fc/platformio_override.ini): 
  - [here](https://github.com/imeszaros/ledclock/blob/e31a81c1ca2a10eb4417e8a380c6760602ac45fc/platformio_override.ini#L50)
  - and [here](https://github.com/imeszaros/ledclock/blob/e31a81c1ca2a10eb4417e8a380c6760602ac45fc/platformio_override.ini#L58)
  - you may need to adjust the power limit [here](https://github.com/imeszaros/ledclock/blob/e31a81c1ca2a10eb4417e8a380c6760602ac45fc/platformio_override.ini#L53)
- in [um_ledclock.h](https://github.com/imeszaros/ledclock/blob/e31a81c1ca2a10eb4417e8a380c6760602ac45fc/wled00/um_ledclock.h) change in the [member initializer list](https://github.com/imeszaros/ledclock/blob/e31a81c1ca2a10eb4417e8a380c6760602ac45fc/wled00/um_ledclock.h#L307) the arguments for `dHoursT`, `dHoursO`, `dMinutesT` and `dMinutesO` from **2** to the desired number of LEDs per segment
- currently it is *not* possible to change the number of LEDs in the separator

**Either if you've changed the number of LEDs or just want to change the order of LEDs, you need to re-map the LEDs [here](https://github.com/imeszaros/ledclock/blob/e31a81c1ca2a10eb4417e8a380c6760602ac45fc/wled00/um_ledclock.h#L325):**
- For example the statement `dHoursT.mapSegment(_7SEG_SEG_A, 6, 7);` maps the 'A' segment of the 'tens' digit to LEDs indexed with `6` and `7`.
- `mapSegment(...)` can be called with variable number of arguments depending on how many LEDs you have per segment.
- `mapSegment(...)` expects LED ordering within a segment:
  - left to right for horizontal segments (e.g. A)
  - top to bottom for vertical segments (e.g. B)

littleferris, 2023-02-01:
so I am doing something different and trying to edit all this, but the sep.map is confusing to me(noob sorry) my led string is a 30 led string so there is only 1 led in each segment and there is 2 leds in the separator. but I dont understand what the numbers in the sep.map means. the leds would be numbers 14 and 15.  This is what I have it look like now, but I dont know what to change sep.map to. would I just change 29/28 to 14 15?

    `void setup() {
        // digit 1
        dHoursT.setShowZero(!hideZero);
        dHoursT.mapSegment(_7SEG_SEG_A, 2);
        dHoursT.mapSegment(_7SEG_SEG_B, 1);
        dHoursT.mapSegment(_7SEG_SEG_C, 6);
        dHoursT.mapSegment(_7SEG_SEG_D, 5);
        dHoursT.mapSegment(_7SEG_SEG_E, 4);
        dHoursT.mapSegment(_7SEG_SEG_F, 3);
        dHoursT.mapSegment(_7SEG_SEG_G, 0);

        // digit 2
        dHoursO.mapSegment(_7SEG_SEG_A, 9);
        dHoursO.mapSegment(_7SEG_SEG_B, 8);
        dHoursO.mapSegment(_7SEG_SEG_C, 13);
        dHoursO.mapSegment(_7SEG_SEG_D, 12);
        dHoursO.mapSegment(_7SEG_SEG_E, 11);
        dHoursO.mapSegment(_7SEG_SEG_F, 10);
        dHoursO.mapSegment(_7SEG_SEG_G, 7);

        sep.map(2,
            2, 0, 29,
            4, 0, 28);

        // digit 3
        dMinutesT.mapSegment(_7SEG_SEG_A, 18);
        dMinutesT.mapSegment(_7SEG_SEG_B, 17);
        dMinutesT.mapSegment(_7SEG_SEG_C, 22);
        dMinutesT.mapSegment(_7SEG_SEG_D, 21);
        dMinutesT.mapSegment(_7SEG_SEG_E, 20);
        dMinutesT.mapSegment(_7SEG_SEG_F, 19);
        dMinutesT.mapSegment(_7SEG_SEG_G, 16);

        // digit 4
        dMinutesO.mapSegment(_7SEG_SEG_A, 25);
        dMinutesO.mapSegment(_7SEG_SEG_B, 24);
        dMinutesO.mapSegment(_7SEG_SEG_C, 29);
        dMinutesO.mapSegment(_7SEG_SEG_D, 28);
        dMinutesO.mapSegment(_7SEG_SEG_E, 27);
        dMinutesO.mapSegment(_7SEG_SEG_F, 26);
        dMinutesO.mapSegment(_7SEG_SEG_G, 23);`

imeszaros, 2023-02-02:
@littleferris You're right, there was no word about the separator in the OP.

The idea is that the 7 segment displays and the separator together create an imaginary two dimensional display. The number of rows and columns of this display depends on how many leds a segment of a 7 segment display has.

For example, if the whole display would only stand of one 7 segment display, its 'pixels' could be addressed like:

```
//     x (0,1..n) x
//   (1,0) x x (1,n+1)
//   (2,0) x x (2,n+1)
//     x (3,1..n) x
//   (4,0) x x (4,n+1)
//   (5,0) x x (5,n+1)
//     x (6,1..n) x
```

The first argument of the separators map function tells how many leds it has (e.g. 2). Then for each LED comes a row and a column index and finally a LED index.

The separator display in my case only has one 'column', so the column index is zero for both LEDs (indices are local to the 'display part'). Since I use 2 leds per segment, my separator is 7 'pixel' high, and I positioned the LEDs above and below the center segment of the 7 segment displays:

```
//     x (0,1..n) x
//   (1,0) x x (1,n+1)
//   (2,0) x x (2,n+1)  here row = 2
//     x (3,1..n) x
//   (4,0) x x (4,n+1)  here row = 4
//   (5,0) x x (5,n+1)
//     x (6,1..n) x

sep.map(2 // 2 leds,
  /*row: */ 2, /* col: */ 0, /* led: */ 29,
  /*row: */ 4, /* col: */ 0, /* led: */ 28);
```

I hope this helps!

DJprasenjit, 2023-02-06:
thanks . i changed the order of leds its working. please anyone tell me how to change time zone stuff in code and auto brightness off it would be helpfull

dipietrofilippo, 2023-05-01:
Hi, I built the 4-display clock with 7 segments and two separators, each segment has 9 LEDs, I tried to change the parameters as you showed in the example, but I couldn't get anything done. this is what i wrote:

 void setup() {
        // digit 1
        dHoursT.setShowZero(!hideZero);
        dHoursT.4mapSegment(_7SEG_SEG_A, 27 , 28 , 29 , 30 , 31 , 32 , 33 , 34 , 35);
        dHoursT.5mapSegment(_7SEG_SEG_B, 36 , 37 , 38 , 39 , 40 , 41 , 42 , 43 , 44);
        dHoursT.7mapSegment(_7SEG_SEG_C, 54 , 55 , 56 , 57 , 58 , 59 , 60 , 61 , 62);
        dHoursT.1mapSegment(_7SEG_SEG_D, 8 , 7 , 6 , 5 , 4 , 3 , 2 , 1 , 0);
        dHoursT.2mapSegment(_7SEG_SEG_E, 17 , 16 , 15 , 14 , 13 , 12 , 11 , 10 , 9);
        dHoursT.3mapSegment(_7SEG_SEG_F, 26 , 25 , 24 , 23 , 22 , 21 , 20 , 19 , 18);
        dHoursT.6mapSegment(_7SEG_SEG_G, 53 , 52 , 51 , 50 , 49 , 48 , 47 , 46 , 45);

        // digit 2
        dHoursT.setShowZero(!hideZero);
        dHoursT.11mapSegment(_7SEG_SEG_A, 90 , 97 , 92 , 93 , 94, 95 , 96 , 97 , 98);
        dHoursT.12mapSegment(_7SEG_SEG_B, 99 , 100 , 101 , 102 , 103 , 104 , 105 , 106 , 107);
        dHoursT.14mapSegment(_7SEG_SEG_C, 117 , 118 , 119 , 120 , 121 , 122 , 123 , 124 , 125);
        dHoursT.8mapSegment(_7SEG_SEG_D, 71 , 70 , 69 , 68 , 67 , 66 , 65 , 64 , 63);
        dHoursT.9mapSegment(_7SEG_SEG_E, 80 , 79 , 78 , 77 , 76 , 75 , 74 , 73 , 72);
        dHoursT.10mapSegment(_7SEG_SEG_F, 89 , 88 , 87 , 86 , 85, 84 , 83 , 82 , 81);
        dHoursT.13mapSegment(_7SEG_SEG_G, 116 , 115 , 114 , 113 , 112 , 111 , 110 , 109 , 108);
       
        // addLed must be called LC_SEP_LEDS times
        sep.addLed(2, 0, 126);
        sep.addLed(4, 0, 127);
      
        // digit 3
        dHoursT.setShowZero(!hideZero);
        dHoursT.18mapSegment(_7SEG_SEG_A, 155 , 156 , 157 , 158 , 159, 160 , 161 , 162 , 163 );
        dHoursT.19mapSegment(_7SEG_SEG_B, 164 , 165 , 166 , 167 , 168 , 169 , 170 , 171 , 172 );
        dHoursT.21mapSegment(_7SEG_SEG_C, 182 , 183 , 184 , 185 , 186 , 187 , 188 , 189 , 190 );
        dHoursT.15mapSegment(_7SEG_SEG_D, 136 , 135 , 134 , 133 , 132 , 131 , 130 , 129 , 128 );
        dHoursT.16mapSegment(_7SEG_SEG_E, 145 , 144 , 143 , 142 , 141, 140 , 139 , 138 , 137 );
        dHoursT.17mapSegment(_7SEG_SEG_F, 154 , 153 , 152 , 151 , 150 , 149 , 148 , 147 , 146 );
        dHoursT.20mapSegment(_7SEG_SEG_G, 181 , 180 , 179 , 178 , 177 , 176 , 175 , 174 , 173 );
       
        // digit 4
        dHoursT.setShowZero(!hideZero);
        dHoursT.25mapSegment(_7SEG_SEG_A, 218 , 219 , 220 , 221 , 222, 223 , 224 , 225 , 226 );
        dHoursT.26mapSegment(_7SEG_SEG_B, 227 , 228 , 229 , 230 , 231 , 232 , 233 , 234 , 235 );
        dHoursT.28mapSegment(_7SEG_SEG_C, 245 , 246 , 247 , 248 , 249 , 250 , 251 , 252 , 253 );
        dHoursT.22mapSegment(_7SEG_SEG_D, 199 , 198 , 197 , 196 , 195, 194 , 193 , 192 , 191 );
        dHoursT.23mapSegment(_7SEG_SEG_E, 208 , 207 , 206 , 205 , 240 , 203 , 202 , 201 , 200 );
        dHoursT.24mapSegment(_7SEG_SEG_F, 217 , 216 , 215 , 214 , 213 , 212 , 211 , 210 , 209 );
        dHoursT.27mapSegment(_7SEG_SEG_G, 244 , 243 , 242 , 241 , 240 , 239 , 238 , 237 , 236 );

grazie mille

aldo-2000, 2023-06-17:
I want to do the same…9 LEDs per segment
Do you have  a solution ?

> Hi, I built the 4-display clock with 7 segments and two separators, each segment has 9 LEDs, I tried to change the parameters as you showed in the example, but I couldn't get anything done. this is [...]

BoobixX, 2023-09-22:
> **If you want to change how many LEDs make up a segment:**
> 
> * change the total number of LEDs in [platformio_override.ini](https://github.com/imeszaros/ledclock/blob/e31a81c1ca2a10eb4417e8a380c6760602ac45fc/platformio_override.ini):
>   
>   * [here](https://github.com/imeszaros/ledclock/blob/e31a81c1ca2a10eb4417e8a380c6760602ac45fc/platformio_override.ini#L50) **_OK_**
>   * and [here](https://github.com/imeszaros/ledclock/blob/e31a81c1ca2a10eb4417e8a380c6760602ac45fc/platformio_override.ini#L58) **_OK_**
>   * you may need to adjust the power limit [here](https://github.com/imeszaros/ledclock/blob/e31a81c1ca2a10eb4417e8a380c6760602ac45fc/platformio_override.ini#L53) **_OK_**
> * in [um_ledclock.h](https://github.com/imeszaros/ledclock/blob/e31a81c1ca2a10eb4417e8a380c6760602ac45fc/wled00/um_ledclock.h) change in the [member initializer list](https://github.com/imeszaros/ledclock/blob/e31a81c1ca2a10eb4417e8a380c6760602ac45fc/wled00/um_ledclock.h#L307) the arguments for `dHoursT`, `dHoursO`, `dMinutesT` and `dMinutesO` from **2** to the desired number of LEDs per segment

_**Problem:** There is not the number_ "2" _but_ "LC_LEDS_PER_SEGM"
[image]

_I modified the file_ "const_ledclock.h _as follows_:
[image]

_It does not work_  !!

> * currently it is _not_ possible to change the number of LEDs in the separator
> 
> **Either if you've changed the number of LEDs or just want to change the order of LEDs, you need to re-map the LEDs [here](https://github.com/imeszaros/ledclock/blob/e31a81c1ca2a10eb4417e8a380c6760602ac45fc/wled00/um_ledclock.h#L325):**
> 
> * For example the statement `dHoursT.mapSegment(_7SEG_SEG_A, 6, 7);` maps the 'A' segment of the 'tens' digit to LEDs indexed with `6` and `7`.
> * `mapSegment(...)` can be called with variable number of arguments depending on how many LEDs you have per segment.
> * `mapSegment(...)` expects LED ordering within a segment:
>   
>   * left to right for horizontal segments (e.g. A)
>   * top to bottom for vertical segments (e.g. B)
> 
> EDIT: about the mapping of the separator, read [this](https://github.com/imeszaros/ledclock/issues/5#issuecomment-1413304444).

_Your configuration with 2 LEDs per segment works very well without anything on line 346_
[image]

Can you help me please ?
THANKS
Francis

cocolino83, 2023-10-14:
ho provato a modificare il codice, ma non riesco a concludere il salvataggio, in quanto su VSC mi da un sempre errore. anche se il codice sorgente non viene modificato.
da sempre errore * Invalid environment name 'codm-controller-0.6-rev2'. The name can contain alphanumeric, 
[image]
underscore, and hyphen characters (a-z, 0-9, -, _)  mi potreste inviare il file inmaggine con la modifica dei numero dei led da 2 a 1 led per segmento?

`void setup() {
    // digit 1
    dHoursT.setShowZero(!hideZero);
    dHoursT.mapSegment(_7SEG_SEG_A, 2);
    dHoursT.mapSegment(_7SEG_SEG_B, 1);
    dHoursT.mapSegment(_7SEG_SEG_C, 6);
    dHoursT.mapSegment(_7SEG_SEG_D, 5);
    dHoursT.mapSegment(_7SEG_SEG_E, 4);
    dHoursT.mapSegment(_7SEG_SEG_F, 3);
    dHoursT.mapSegment(_7SEG_SEG_G, 0);

    // digit 2
    dHoursO.mapSegment(_7SEG_SEG_A, 9);
    dHoursO.mapSegment(_7SEG_SEG_B, 8);
    dHoursO.mapSegment(_7SEG_SEG_C, 13);
    dHoursO.mapSegment(_7SEG_SEG_D, 12);
    dHoursO.mapSegment(_7SEG_SEG_E, 11);
    dHoursO.mapSegment(_7SEG_SEG_F, 10);
    dHoursO.mapSegment(_7SEG_SEG_G, 7);

    sep.map(2,
        2, 0, 29,
        4, 0, 28);

    // digit 3
    dMinutesT.mapSegment(_7SEG_SEG_A, 18);
    dMinutesT.mapSegment(_7SEG_SEG_B, 17);
    dMinutesT.mapSegment(_7SEG_SEG_C, 22);
    dMinutesT.mapSegment(_7SEG_SEG_D, 21);
    dMinutesT.mapSegment(_7SEG_SEG_E, 20);
    dMinutesT.mapSegment(_7SEG_SEG_F, 19);
    dMinutesT.mapSegment(_7SEG_SEG_G, 16);

    // digit 4
    dMinutesO.mapSegment(_7SEG_SEG_A, 25);
    dMinutesO.mapSegment(_7SEG_SEG_B, 24);
    dMinutesO.mapSegment(_7SEG_SEG_C, 29);
    dMinutesO.mapSegment(_7SEG_SEG_D, 28);
    dMinutesO.mapSegment(_7SEG_SEG_E, 27);
    dMinutesO.mapSegment(_7SEG_SEG_F, 26);
    dMinutesO.mapSegment(_7SEG_SEG_G, 23);`

BoobixX, 2023-10-18:
> ho provato a modificare il codice, ma non riesco a concludere il salvataggio, in quanto su VSC mi da un sempre errore. anche se il codice sorgente non viene modificato. da sempre errore * Invalid [...]

Hi, you can replace - with _ in lines 605 and 614. Compilation will be possible but I don't know the impact in the program.
If a charitable soul would like to respond to my post above for the number of LEDs
Thanks

Francois

loverboyritesh, 2023-10-20:
> I want to do the same…9 LEDs per segment Do you have a solution ?

I think there shouldnt be any space between the number and comma like - dHoursT.11mapSegment(_7SEG_SEG_A, 90, 97, 92, 93, 94, 95, 96, 97, 98);

i may be wrong . but try .

lucky-julie99, 2023-12-22:
Hello, is this solution for using a different number of LEDs still up to date?

I cloned the repo in VS an wanted to implement the changes for using 3 LEDs, but there are some completely different files and the part where the order of the LEDs is written, is not there at all.
So then I downloaded the zip file and here everything looks like described.
Which version should I use now?

And second question, can someone explain to me, how to properly build the program using VS and PlatformIO? This is my first project and I don't know what I'm doing wrong. I load the ledclock folder in VS Studio, then I click PlatformIO:build and afterwards PlatformIO:Upload which both work. But if I change for example the LED Pin inside the platformio_override.ini file, nothing happens and it still uses Pin 4 like before. 

Thank you for helping and kind regards, Lucky

imeszaros, 2023-12-22:
Hi there! The top of the main branch contains experimental code migrated to the latest WLED release. Very soon I'm planning to release another version of the firmware (as soon is blocking issues are fixed in WLED). That is why you're seeing different files, there were some big refactorings on the WLED side and I needed to adapt my code.

You can however use the latest stable release to apply your changes as described in this issue. In your repo, create a new branch on tag `ledclock-1.2.0`.

I hope this helps!

lucky-julie99, 2023-12-22:
Hi, thank you very much for that fast response :)

With the 1.2.0 I can open the project in VS, build and upload it with PlatformIO and it runs (but without the Wi-Fi even though it is configured) . And when I change the LED Port in platformio_override.ini, build and upload again, nothing changes. It still uses PIN 4 like before. The PIN isn't that important, it is just to see if the change works. But I need to set se DEFAULT_LED_COUNT to 86, which therefor also doesn't work. 
I looks like the changes I make in platform_override.ini and the other header files are not used in the build and I really can't figure out why. I never worked with PlatformIO or ESP32 programming before, so hopefully you can help me, if I do something completely wrong.

imeszaros, 2023-12-22:
I'm glad building and upload works! :) In theory changing those defines in `platformio_override.ini` should do the trick.. But those are really just default values. When WLED boots for the first time, it will create a configuration file which will contain the LED settings, changing the defaults afterwards won't make a difference.

Try to erase the ESP32 flash storage and upload it again to see what settings it will initialize with.
[image]

lucky-julie99, 2023-12-29:
Thank you very much, that worked.
Really nice project :)

TechHamara, 2024-01-07:
> Hi, you can replace - with _ in lines 605 and 614. Compilation will be possible but I don't know the impact in the program. If a charitable soul would like to respond to my post above for the number [...]

>> > Hi, i have replace [env:codm_ontroller_0.6] and [env:codm_controller_0.6_rev2] to
 . with _ in lines 605 and 614. to work properly.

TechHamara, 2024-01-07:
hi [imeszaros](https://github.com/imeszaros)
can i build for esp8266 board.
if possible then how can i build. i will but only esp32 release.

imeszaros, 2024-01-31:
> hi [imeszaros](https://github.com/imeszaros) can i build for esp8266 board. if possible then how can i build. i will but only esp32 release.

ESP8266 is not supported by this project.

tadejp12, 2024-01-31:
Hi, i am completely new to esp32 and all of its stuff. I installed this project on esp, but i have just 1 led diode for each segment. I already changed and configured files for led positioning as you described, but now i do not know how to save all these files to format which i could put to esp32. Thank you.

cocolino83, 2024-02-01:
HI. I downloaded the new update. I've tried and tried but I can't get my watch to work properly. I modified the code for 1 segment led, but I don't understand where I'm wrong in the code. I'm sending you my images hoping you can give me a hand. thanks 1000 in advance
[image]
[image]
[image]
[image]
[image]

imeszaros, 2024-02-01:
Hi @cocolino83 ,

your `LC_LEDMAP` looks good already.

You need to change `LC_SEP_LED_ROWS` to `1, 3` since your separator LEDs are in rows 1 and 3 instead of 2 and 4 (row numbers starting from zero!).

Additionally, as the compiler says, you need to change `DEFAULT_LED_COUNT` in `platformoi_override.ini` to `65`.

Compile and upload, and do a factory reset in WLED to erase the existing LED settings and apply the new ones.

imeszaros, 2024-02-01:
> Hi, i am completely new to esp32 and all of its stuff. I installed this project on esp, but i have just 1 led diode for each segment. I already changed and configured files for led positioning as [...]

Hi, you need to open the project in VSCode / PlatformIO and compile it. You can also use it to upload the code to the ESP32. Look for PlatformIO tutorials if you're new to it.

cocolino83, 2024-02-01:
> CIAO@cocolino83,
> 
> stai `LC_LEDMAP`già bene.
> 
> È necessario passare `LC_SEP_LED_ROWS`a `1, 3`poiché i LED separatori si trovano nelle righe 1 e 3 anziché in 2 e 4 (numeri di riga che iniziano da zero!).
> 
> Inoltre, come dice il compilatore, è necessario passare `DEFAULT_LED_COUNT`a .`platformoi_override.ini``65`
> 
> Compila e carica ed esegui un ripristino delle impostazioni di fabbrica in WLED per cancellare le impostazioni LED esistenti e applicare quelle nuove.

thank you 1000 everything went well. 
[image]

I INSERT THE LINK FOR THE FILE.BIN 1 ONLY LED EACH SEGMENT. I BELIEVE IT MAY BE USEFUL TO OTHERS.
https://github.com/cocolino83/Ledclook-1.3.0

Nerdish22, 2024-02-05:
Hello Y'all.   Im new to all this.  trying to add/chang led count.   I understand how to do that. but to get to the road to change that's on my esp 32 is where I'm lost. I have wled clock running and on a led strip to test. like I sed I'm completly new.  just need some guidence   thanks

cocolino83, 2024-02-05:
> Ciao a tutti. Sono nuovo a tutto questo. cercando di aggiungere/modificare il conteggio dei led. Capisco come farlo. ma per arrivare alla strada del cambiamento che è sul mio ESP 32 è dove mi sono perso. Ho controllato l'orologio in funzione e ho testato una striscia LED. come se fossi completamente nuovo. ho solo bisogno di qualche dritta, grazie

Quanti led per segmento?

Nerdish22, 2024-02-05:
going to go from 2 led per seg to 5.    I understand on what he's telling me to do.   just not how to get the coat and to upload it

cocolino83, 2024-02-05:
> passerò da 2 led per seg a 5. Ho capito cosa mi dice di fare. semplicemente non come prendere il cappotto e caricarlo

Quindi devi solo caricare il firmware.bin e già configurato per due led per segmento. Per caricarlo su esp32 ti occorre il tool. Prova con questo. https://github.com/hanhdt/esp32-flash-tool

Nerdish22, 2024-02-05:
> Quindi devi solo caricare il firmware.bin e già configurato per due led per segmento. Per caricarlo su esp32 ti occorre il tool. Prova con questo. https://github.com/hanhdt/esp32-flash-tool

where do I get the bin from?  do I download the hole Zip and get to it that way or copy past    I'm more of a hands on guy

cocolino83, 2024-02-05:
> > > passerò da 2 led per seg a 5. Ho capito cosa mi dice di fare. semplicemente non come prendere il cappotto e caricarlo
> > 
> > 
> > Quindi devi solo caricare il firmware.bin e già configurato per due led per segmento. Per caricarlo su esp32 ti occorre il tool. Prova con questo. https://github.com/hanhdt/esp32-flash-tool
> 
> da dove prendo il cestino? faccio a scaricare il buco Zip e ad arrivarci in quel modo o lo copio? Sono più un tipo pratico

Non devi scaricare il file zip ma firmware.bin , una volta scaricato apri il tool per flash esp collegarlo il tuo esp 32 e al PC e installalo. Comunque ti consiglio di guardare su YouTube " wled esp32" 👍

Nerdish22, 2024-02-05:
> Non devi scaricare il file zip ma firmware.bin , una volta scaricato apri il tool per flash esp collegarlo il tuo esp 32 e al PC e installalo. Comunque ti consiglio di guardare su YouTube " wled [...]

Thanks I'm doing it right now.  hope it helps me out

JureqB, 2024-02-07:
Hello, I don't know anything about programming, barely understand Platformio but I wanted to change the number of LEDs from 2 to 3 per segment in version 1.3.0
So according to the instructions I change in const_ledclock.h LC_LEDS_PER_SEGM 3, change the entire matrix accordingly, then correct the position of LC_SEP_LED_ROWS 3, 5 and enter -D DEFAULT_LED_COUNT=189 in platformio_override.ini
It seems to me that there is nothing else to set, but when compiling it shows me errors:

Compiling .pio\build\esp32dev\libd91\FS\FS.cpp.o
In file included from wled00/const.h:4:0,
                 from wled00/wled.h:176,
                 from wled00/set.cpp:1:
wled00/const_ledclock.h:100:136: note: #pragma message: Macro DEFAULT_LED_COUNT is not defined or is not equal to the calculated number of total LEDs: ((((3 + 2) * 4) + 1) * ((3 * 2) + 3))
 #pragma message "Macro DEFAULT_LED_COUNT is not defined or is not equal to the calculated number of total LEDs: " LC_XSTR(LC_TOTAL_LEDS)

 ^
wled00/const_ledclock.h:99:20: note: in definition of macro 'LC_STR'
 #define LC_STR(x) #x
                    ^
wled00/const_ledclock.h:100:115: note: in expansion of macro 'LC_XSTR'
 #pragma message "Macro DEFAULT_LED_COUNT is not defined or is not equal to the calculated number of total LEDs: " LC_XSTR(LC_TOTAL_LEDS)
                                                                                                                   ^
wled00/set.cpp: In function 'void handleSettingsSet(AsyncWebServerRequest*, byte)':
wled00/set.cpp:720:103: error: 'dynamic_cast' not permitted with -fno-rtti
     LedClockSettings* settings = dynamic_cast<LedClockSettings *>(usermods.lookup(USERMOD_ID_LEDCLOCK));
                                                                                                       ^
*** [.pio\build\esp32dev\src\set.cpp.o] Error 1
In file included from wled00/const.h:4:0,
                 from wled00/wled.h:176,
                 from wled00/wled.cpp:2:
wled00/const_ledclock.h:100:136: note: #pragma message: Macro DEFAULT_LED_COUNT is not defined or is not equal to the calculated number of total LEDs: ((((3 + 2) * 4) + 1) * ((3 * 2) + 3))
 #pragma message "Macro DEFAULT_LED_COUNT is not defined or is not equal to the calculated number of total LEDs: " LC_XSTR(LC_TOTAL_LEDS)

 ^
wled00/const_ledclock.h:99:20: note: in definition of macro 'LC_STR'
 #define LC_STR(x) #x
                    ^
wled00/const_ledclock.h:100:115: note: in expansion of macro 'LC_XSTR'
 #pragma message "Macro DEFAULT_LED_COUNT is not defined or is not equal to the calculated number of total LEDs: " LC_XSTR(LC_TOTAL_LEDS)
                                                                                                                   ^
In file included from wled00/const.h:4:0,
                 from wled00/wled.h:176,
                 from C:/Users/jtp/Downloads/ledclock-ledclock-1.3.0/ledclock-ledclock-1.3.0/wled00/wled00.ino:13:
wled00/const_ledclock.h:100:136: note: #pragma message: Macro DEFAULT_LED_COUNT is not defined or is not equal to the calculated number of total LEDs: ((((3 + 2) * 4) + 1) * ((3 * 2) + 3))
 #pragma message "Macro DEFAULT_LED_COUNT is not defined or is not equal to the calculated number of total LEDs: " LC_XSTR(LC_TOTAL_LEDS)

 ^
wled00/const_ledclock.h:99:20: note: in definition of macro 'LC_STR'
 #define LC_STR(x) #x
                    ^
wled00/const_ledclock.h:100:115: note: in expansion of macro 'LC_XSTR'
 #pragma message "Macro DEFAULT_LED_COUNT is not defined or is not equal to the calculated number of total LEDs: " LC_XSTR(LC_TOTAL_LEDS)
                                                                                                                   ^
In file included from wled00/const.h:4:0,
                 from wled00/wled.h:176,
                 from wled00/wled_serial.cpp:1:
wled00/const_ledclock.h:100:136: note: #pragma message: Macro DEFAULT_LED_COUNT is not defined or is not equal to the calculated number of total LEDs: ((((3 + 2) * 4) + 1) * ((3 * 2) + 3))
 #pragma message "Macro DEFAULT_LED_COUNT is not defined or is not equal to the calculated number of total LEDs: " LC_XSTR(LC_TOTAL_LEDS)

 ^
wled00/const_ledclock.h:99:20: note: in definition of macro 'LC_STR'
 #define LC_STR(x) #x
                    ^
wled00/const_ledclock.h:100:115: note: in expansion of macro 'LC_XSTR'
 #pragma message "Macro DEFAULT_LED_COUNT is not defined or is not equal to the calculated number of total LEDs: " LC_XSTR(LC_TOTAL_LEDS)
                                                                                                                   ^
In file included from wled00/const.h:4:0,
                 from wled00/wled.h:176,
                 from wled00/ws.cpp:1:
wled00/const_ledclock.h:100:136: note: #pragma message: Macro DEFAULT_LED_COUNT is not defined or is not equal to the calculated number of total LEDs: ((((3 + 2) * 4) + 1) * ((3 * 2) + 3))
 #pragma message "Macro DEFAULT_LED_COUNT is not defined or is not equal to the calculated number of total LEDs: " LC_XSTR(LC_TOTAL_LEDS)

 ^
wled00/const_ledclock.h:99:20: note: in definition of macro 'LC_STR'
[... 14 lines cut ...]
                    ^
wled00/const_ledclock.h:100:115: note: in expansion of macro 'LC_XSTR'
 #pragma message "Macro DEFAULT_LED_COUNT is not defined or is not equal to the calculated number of total LEDs: " LC_XSTR(LC_TOTAL_LEDS)
                                                                                                                   ^
[... cut ...]
wled00/xml.cpp: In function 'void getSettingsJS(byte, char*)':
wled00/xml.cpp:770:103: error: 'dynamic_cast' not permitted with -fno-rtti
     LedClockSettings* settings = dynamic_cast<LedClockSettings *>(usermods.lookup(USERMOD_ID_LEDCLOCK));
                                                                                                       ^
*** [.pio\build\esp32dev\src\xml.cpp.o] Error 1
===================================================== [FAILED] Took 15.97 seconds 

Can I ask you to explain where I made a mistake ?

semiaquatic, 2024-02-07:
Hi, please double check whether `DEFAULT_LED_COUNT=189` is set in `platformio_override.ini`, and youre compiling against the `ledclock` environment:

[image]

imeszaros, 2024-02-07:
Hi, please double check whether `DEFAULT_LED_COUNT=189` is set in `platformio_override.ini`, and youre compiling against the `ledclock` environment:

[image]

JureqB, 2024-02-07:
It seems like I'm doing everything correctly
[image]

imeszaros, 2024-02-07:
As I indicated on the screenshot, the target environment should be "ledclock". Yours is "esp32dev" as the statusbar is showing it. Click on it and change it to "ledclock".

Nerdish22, 2024-02-10:
Some help would be nice.  I'm getting the led count not right.   I'm I missing something? if my math is right I'm thinking its 377 and I have that set.   

[image]

Nerdish22, 2024-02-10:
Im not sure y I'm getting this. anyone able to help?

[image]

JureqB, 2024-02-10:
I see that You have **LC_LEDS_PER_SEGM 2** instead of **5**

imeszaros, 2024-02-10:
@JureqB is right in all points! Additionally, LED 5 and 6 is not in the matrix at all. After LED 4 LED 7 is coming at the bottom left corner of the first digit.. Is that intentional?

Nerdish22, 2024-02-10:
> I see that:
> 
> * You have **LC_LEDS_PER_SEGM 2** instead of **5**
> * instead of 105 it should be -1 and the rest corrected
> * you unnecessarily modified the #define LC_COLS and #define LC_ROWS code
>   ![lc1e3](https://private-user-images.githubusercontent.com/25904878/303835416-28d92766-cb38-43b7-ad26-9e47e7d622d7.jpg?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MDc1ODI3MDcsIm5iZiI6MTcwNzU4MjQwNywicGF0aCI6Ii8yNTkwNDg3OC8zMDM4MzU0MTYtMjhkOTI3NjYtY2IzOC00M2I3LWFkMjYtOWU0N2U3ZDYyMmQ3LmpwZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDAyMTAlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwMjEwVDE2MjY0N1omWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTcxYTAyZmQ0NjhlYTljMTg0MjdlNDhlZGYzZDRmOWZkNjQyOTg1YWUzMjQ5ZTYyMjdkZDdlMGI3NjRiNDhmODYmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.aKurmb55srZp5bnBj8CspJ51PGrggAwWfOkRqPX9mNc)

wow thanks JureqB.   I was running on 26+ hours no sleep. I can't believe I did that. I know a fresh pair od eyes will help.   let me chang that and see what happens

Nerdish22, 2024-02-10:
> @JureqB is right in all points! Additionally, LED 5 and 6 is not in the matrix at all. After LED 4 LED 7 is coming at the bottom left corner of the first digit.. Is that intentional?

Opps Long nights no sleep.  can't count haha

Nerdish22, 2024-02-10:
> I see that:
>
> * You have **LC_LEDS_PER_SEGM 2** instead of **5**
> * instead of 105 it should be -1 and the rest corrected
> * you unnecessarily modified the #define LC_COLS and #define LC_ROWS code [...]

/
/
/jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MDc1ODc3MDEsIm5iZiI6MTcwNzU4NzQwMSwicGF0aCI6Ii8yNTkwNDg3OC8zMDM4MzU0MTYtMjhkOTI3NjYtY2IzOC00M2I3LWFkMjYtOWU0N2U3ZDYyMmQ3LmpwZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDAyMTAlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwMjEwVDE3NTAwMVomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTU2Njk5YjQ2MDE3N2Y2N2ExYjhkNTVhZTRiZjdmYTA4YjZhMzgyNjA1Y2QyZDRjMmJmMWQzYTAwNDcwZjczMjkmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.guVihVMqc1cybp7bLLymHww4Fkn3wQBVsb_x96rYJbE)
[image]

Nerdish22, 2024-02-10:
> I see that:
>
> * You have **LC_LEDS_PER_SEGM 2** instead of **5**
> * instead of 105 it should be -1 and the rest corrected
> * you unnecessarily modified the #define LC_COLS and #define LC_ROWS code [...]

[image]

now I have another problem.    led 1-5 only the first 4 are working.  and on the sep leds I have 5 between them not I'm understating everything right should not be that way

veverica12345, 2024-02-18:
Hello, can someone send me the code for 4 LED diodes per segment?

veverica12345, 2024-02-20:
I don't have wled on the board anymore, I rebooted, I have a problem when I make changes in the code

[image]
[image]

veverica12345, 2024-02-20:
I have to change the order of the diodes in the segment because it's not the same as in this project, so I'll see what happens next.
[image]

cocolino83, 2024-02-20:
> Devo cambiare l'ordine dei diodi nel segmento perché non è lo stesso di questo progetto, quindi vedrò cosa succede dopo. ![Snimak ekrana 2024-02-20 120658](https://private-user-images.githubusercontent.com/151787209/306238701-376b39f0-a11e-445f-a315-c0d8d8468a7f.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MDg0Mjc2NDIsIm5iZiI6MTcwODQyNzM0MiwicGF0aCI6Ii8xNTE3ODcyMDkvMzA2MjM4NzAxLTM3NmIzOWYwLWExMWUtNDQ1Zi1hMzE1LWMwZDhkODQ2OGE3Zi5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQwMjIwJTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MDIyMFQxMTA5MDJaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT1hYTJkYThmYzYxZjIzMTA2MDJlODY0OWNiNzMyNGVjYTQ2MTRmMzBjZTBhODc4NmYwZDc0NDQ2NDNjMWZiOGRiJlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCZhY3Rvcl9pZD0wJmtleV9pZD0wJnJlcG9faWQ9MCJ9.xWg_A5QVWAEpQDhSZ3g9Af8ljQWjfCVbzxWanEMl1n0)

I too had this problem. but after uploading it fixed itself. anyway, that's how I do it. once I click on the upload arrow, I wait for it to finish, but without connecting the esp32. I go to the '' pio\build\ledclock\ '' folder, take the.bin file and load this.bin file directly, from the ledclock interface '' config-security and updates-manual ota update.'' I select the ledclok bin file, after it is loaded you have to go to the main screen ledclok info and reboot wled, it restarts and if all the code is correct it works. I hope you can understand. HI

veverica12345, 2024-02-20:
If you think this way, it doesn't work

[image]

cocolino83, 2024-02-20:
> Se la pensi in questo modo, non funziona
> 
> ![Snimak ekrana 2024-02-20 183656](https://private-user-images.githubusercontent.com/151787209/306364124-66c8afeb-421a-471e-91fa-191f0691e107.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MDg0NTY3ODEsIm5iZiI6MTcwODQ1NjQ4MSwicGF0aCI6Ii8xNTE3ODcyMDkvMzA2MzY0MTI0LTY2YzhhZmViLTQyMWEtNDcxZS05MWZhLTE5MWYwNjkxZTEwNy5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQwMjIwJTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MDIyMFQxOTE0NDFaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT00ZGY3NzU1NDhlM2EyZjdmNDNmMjA1ZjhkMDFhOGQ4YjdkMTExMGZhMWY3NzQ1NmJiMTcwZWIwMDZlYzk3YjRkJlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCZhY3Rvcl9pZD0wJmtleV9pZD0wJnJlcG9faWQ9MCJ9.emaNfUT4AqeSMwW638iYbOzOO8S9iHipi5FicLmigrI)

have you seen if it created the.bin file in the folder you linked to studio code "the folder you are editing the wled clock 1.3.0 code." pio\build\ledclock.  ?  no in studio code.  but the physical folder.  I'm not on the PC right now.  otherwise I would have sent you some pictures.

veverica12345, 2024-02-20:
If you're on the computer, please send me a few pictures, I think I'm going in the wrong direction, thank you.

veverica12345, 2024-02-20:
> '' config-security and updates-manual ota update.'

I managed to change the number of diodes as you wrote to load directly the firmware.bin file in the ledclock interface "config-security and updates-manual ota update." but now when I want to configure the layout of the diodes, I tried but again an error

veverica12345, 2024-02-20:
[image]

veverica12345, 2024-02-20:
[image]

what is the problem here now?

cocolino83, 2024-02-21:
> ![Snimak ekrana 2024-02-20 225254](https://private-user-images.githubusercontent.com/151787209/306432282-52d3c2c2-7534-4469-98f4-63d96e2100af.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MDg0ODY1MjYsIm5iZiI6MTcwODQ4NjIyNiwicGF0aCI6Ii8xNTE3ODcyMDkvMzA2NDMyMjgyLTUyZDNjMmMyLTc1MzQtNDQ2OS05OGY0LTYzZDk2ZTIxMDBhZi5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQwMjIxJTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MDIyMVQwMzMwMjZaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT1mYzNjMTgwZDViY2VhMzQ1NDk4MjI5OThkMWVhMDljNDhhMDY2YmEyMDZlOWU0MmE1NTIyMjg0M2FlMTkxODdhJlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCZhY3Rvcl9pZD0wJmtleV9pZD0wJnJlcG9faWQ9MCJ9.82CerCuEiZyB-t2r6CG6bpRDKH3IvPxDeO58REsgedc)
> 
> ![Snimak ekrana 2024-02-20 231647](https://private-user-images.githubusercontent.com/151787209/306431981-b729e045-7379-4217-9a43-d466ed68c77c.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MDg0ODY1MjYsIm5iZiI6MTcwODQ4NjIyNiwicGF0aCI6Ii8xNTE3ODcyMDkvMzA2NDMxOTgxLWI3MjllMDQ1LTczNzktNDIxNy05YTQzLWQ0NjZlZDY4Yzc3Yy5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQwMjIxJTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MDIyMVQwMzMwMjZaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT0xZDZmMWIyYzU5YjViMDEyZTg1YjQyMTBiNGNhNDIzYzFlYzc3M2FiNmE5M2E1YTMxMmUxZTdjZjAxYWQ4NTQ1JlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCZhY3Rvcl9pZD0wJmtleV9pZD0wJnJlcG9faWQ9MCJ9.xaN88qzUU5pPQiyOzfpg-YspBHqL09u4h3lhmgnAGLo)
> 
> qual è il problema qui adesso?

but why is the first LED (0) in your watch in the central segment?  Can you show me the electrical connections you made?
[image]

[image]

veverica12345, 2024-02-21:
I drank differently, look at the picture.

[image]

cocolino83, 2024-02-21:
> Ho bevuto diversamente, guarda la foto.
> 
> ![Screenshot_20240221_132113_edit_119859411986396](https://private-user-images.githubusercontent.com/151787209/306622097-6afa4492-4bd3-4615-bfec-06e23bd642b0.jpg?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MDg1MjYwMTcsIm5iZiI6MTcwODUyNTcxNywicGF0aCI6Ii8xNTE3ODcyMDkvMzA2NjIyMDk3LTZhZmE0NDkyLTRiZDMtNDYxNS1iZmVjLTA2ZTIzYmQ2NDJiMC5qcGc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQwMjIxJTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MDIyMVQxNDI4MzdaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT0zZWE5MjJlN2M4NTFjNTA1NDY5Y2Q2Y2EyMzgwNTBlNTBkYTZjNjgxNDcxOTJiNjQwZmNhYjgwOTQ0MmI0YmJiJlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCZhY3Rvcl9pZD0wJmtleV9pZD0wJnJlcG9faWQ9MCJ9.EPt4zmaQ65NtuC5vAAAKe2tDhpCrC4rGOcJmCZsoEFs)
> 
> ![Screenshot_20240221_133526_edit_120261801462377](https://private-user-images.githubusercontent.com/151787209/306623904-8227df97-054b-4a0a-bafa-3a809b03b599.jpg?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MDg1MjYwMTcsIm5iZiI6MTcwODUyNTcxNywicGF0aCI6Ii8xNTE3ODcyMDkvMzA2NjIzOTA0LTgyMjdkZjk3LTA1NGItNGEwYS1iYWZhLTNhODA5YjAzYjU5OS5qcGc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQwMjIxJTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MDIyMVQxNDI4MzdaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT05YjVhYWJiOGNiN2NhODVmNWM1YzRlMzE1YTVlOGZmOWI2ZTg5YWU3OTg5NWYwYjMyYjZjMjkxNGU4NzdlOWM0JlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCZhY3Rvcl9pZD0wJmtleV9pZD0wJnJlcG9faWQ9MCJ9.gx2tGRXMWaj8j99GFohjqGHi2SyFYdHhCCeLo7pQrtk)

https://github.com/imeszaros/ledclock/issues/5#issuecomment-1956550442

I created the firmware.bin for you with your led settings.  you can find it here https://github.com/cocolino83/Led4 I also included the tool that I use for installations on esp.  hope it works for you.  I recommend that you restart the wled from the settings.  HI

veverica12345, 2024-02-21:
[image]

veverica12345, 2024-02-22:
Yes, that's what I did, I installed wled, then I loaded the Firmware file, please check if the cod is correct, I also flashed my controller, before installing wled

cocolino83, 2024-02-22:
> @littleferrisHai ragione, non c'era alcuna parola sul separatore nell'OP.
> 
> L'idea è che i display a 7 segmenti e il separatore creino insieme un display bidimensionale immaginario. Il numero di righe e colonne di questo display dipende da quanti led ha un segmento di un display a 7 segmenti.
> 
> Ad esempio, se l'intero display fosse costituito da un solo display da 7 segmenti, i suoi "pixel" potrebbero essere indirizzati come:
> 
> ```
> //     x (0,1..n) x
> //   (1,0) x x (1,n+1)
> //   (2,0) x x (2,n+1)
> //     x (3,1..n) x
> //   (4,0) x x (4,n+1)
> //   (5,0) x x (5,n+1)
> //     x (6,1..n) x
> ```
> 
> Il primo argomento della funzione mappa separatori indica quanti led ha (es. 2). Quindi per ciascun LED viene fornito un indice di riga e di colonna e infine un indice di LED.
> 
> Il display separatore nel mio caso ha solo una "colonna", quindi l'indice della colonna è zero per entrambi i LED (gli indici sono locali alla "parte del display"). Poiché utilizzo 2 LED per segmento, il mio separatore è alto 7 "pixel" e ho posizionato i LED sopra e sotto il segmento centrale dei display a 7 segmenti:
> 
> ```
> //     x (0,1..n) x
> //   (1,0) x x (1,n+1)
> //   (2,0) x x (2,n+1)  here row = 2
> //     x (3,1..n) x
> //   (4,0) x x (4,n+1)  here row = 4
> //   (5,0) x x (5,n+1)
> //     x (6,1..n) x
> 
> sep.map(2 // 2 leds,
>   /*row: */ 2, /* col: */ 0, /* led: */ 29,
>   /*row: */ 4, /* col: */ 0, /* led: */ 28);
> ```
> 
> Spero che aiuti!

hi I hope you can take a look at the code. I would like to help build his watch at (veverica12345) I tried to create the.bin file for him, but I think I made a mistake and the LEDs don't light up correctly, I'll show you the code I modified for him. total LEDs DEFAULT_LED_COUNT=275
I think I'm wrong in numbering #define LC_SEP_LED_ROWS 4, 8 because I can't understand this well. A thousand thanks

[image]

I found this too. this error too even though I haven't changed anything else

[image]

cocolino83, 2024-02-25:
> ![IMG_20240225_130402_edit_13106379944353](https://private-user-images.githubusercontent.com/151787209/307589246-6c3cc52d-a59d-4a75-a738-5744b2977b4f.jpg?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MDg4NjQyNTgsIm5iZiI6MTcwODg2Mzk1OCwicGF0aCI6Ii8xNTE3ODcyMDkvMzA3NTg5MjQ2LTZjM2NjNTJkLWE1OWQtNGE3NS1hNzM4LTU3NDRiMjk3N2I0Zi5qcGc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQwMjI1JTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MDIyNVQxMjI1NThaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT01ZGY3ZTZkMGM3YzMwNWM1MjVmNmNiZTExMmJkYTRkZjI3ZDBjYWY4OWQ0ZWRjYzNhZDBhYzdmNTUzNjI1MGVkJlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCZhY3Rvcl9pZD0wJmtleV9pZD0wJnJlcG9faWQ9MCJ9.R83gicBSTP5JTrP7AkBg9QxQ6zdZDmrquNn2mGPxoqY)
> 
> I succeeded with the code, now it works fine. Thank you all for your help

I'm glad you managed it 👍.  for curiosity!  Can you show me the code you modified?  I would like to understand where I went wrong in editing it.  HI

veverica12345, 2024-02-25:
[image]
[image]
[image]

[5 more replies]