imeszaros/ledclock #8: Dynamic Cast Not Permitted

Bluedoggo, 2023-01-19:
Receiving this error each time I attempt to compile using VScode

*** [.pio/build/esp32dev/src/set.cpp.o] Error 1
wled00/xml.cpp: In function 'void getSettingsJS(byte, char*)':
wled00/xml.cpp:675:103: error: 'dynamic_cast' not permitted with -fno-rtti
     LedClockSettings* settings = dynamic_cast<LedClockSettings *>(usermods.lookup(USERMOD_ID_LEDCLOCK));

imeszaros, 2023-01-19:
In `platformio_override.ini` there is a line:
```
build_unflags = ${common.build_unflags} -fno-rtti
```
which should actually enable run-time type information to be able to cast dynamically.

It looks like your compiler does not respect this setting. Did you modify anything there?

Bluedoggo, 2023-01-19:
When I attempted to compile the code downloaded from the github repo, without modifying any code, I received the same error.

imeszaros, 2023-01-20:
Is your platformio & toolchain up to date? Other than that, I'm out of ideas, sorry.

DJprasenjit, 2023-02-05:
compiling done upload on esp wroom 32 . but its reboots

closed 2023-03-15 by imeszaros