imeszaros/ledclock #9: esp8266 veriant working but when open clock settings esp reboots

DJprasenjit, 2023-02-05:
### What happened?

i downloaded the latest code of ledclock-esp8266 compiled from platformio and uploaded on nodeMcu working good. but cant set my timezone. when i am going clock setting option esp reboots. and the web server opens on only the wled app. in web browser when i giving the ip somthing starts downloading. please tell me a solutoin to fix my time zone. 
for your info this same code works fine on esp32.
but i wanted it to esp8266.
please reply with a fix

### To Reproduce Bug

platform io compiles error but fix in two three attempts and uploaded with platform io. ledclock esp8266 working everything. i managed to fix the led order. but in esp8266 reboots only when i open the clock settings and time macros option.

### Expected Behavior

in esp32 same programme working fine . i can open clock settings and time macros

### Install Method

Self-Compiled

### What version of WLED?

lestest build 1.2.0

### Which microcontroller/board are you seeing the problem on?

ESP8266

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

imeszaros, 2023-02-05:
Sorry, I can't helo with ESP8266. My project was never meant to work with it, I do not have an ESP8266 board to try it out. If you have multiple boards you should try to use another one. Other guys did not have this problem as far as i'm aware.

DJprasenjit, 2023-02-06:
can you tell me where is  the manual settings for the clock setting time zone and auto brightness settings? if i do it in the code. then the clock work fine. i have to set time zone IST india and auto brightness off

imeszaros, 2023-02-18:
Timezone setting is loaded here: https://github.com/imeszaros/ledclock/blob/master/wled00/cfg.cpp#L360
Auto brightness setting can be overridden here: https://github.com/imeszaros/ledclock/blob/master/wled00/um_ledclock.h#L777