imeszaros/ledclock #10: Is there a way to change default effect profile?

sbeerda, 2023-02-17:
Looking to change the default powerup effect. I'm assuming it would be in the gui under led setting but that is disabled, is there a way to enable it?

Thanks in advance.

imeszaros, 2023-02-18:
Yes, LED settings are disabled, since the project is designed to a specific hardware. However, you can easily enable it:
- install NodeJS
- install PlatformIO
- checkout the repository
- open the code in platformio
- remove the commenting in https://github.com/imeszaros/ledclock/blob/master/wled00/data/settings.htm around line 33
- in the project directory, run `npm run build`
- build the project with PlatformIO
- upload with PlatformIO

Let me know whether this worked. I'll consider to put this setting back in a future version.

sbeerda, 2023-02-19:
I removed the commenting around line 33 for the LED preferences, erased flash on esp32 and uploaded again. No joy.

using VS Code on Mac

sbeerda, 2023-03-05:
Took me a little bit to figure it out, but I managed to do the `npm run build` and it works!!! Thanks again @imeszaros 👏

closed 2023-03-05 by imeszaros