imeszaros/ledclock #17: Number of Leds greater than 255 (byte overflow?)

Zaraxxx, 2024-02-12:
### What happened?

Hello, I have made a custom version of ledclock using led strips that have been cut, I have 9 leds per segments.

I can build,publish and update my esp32 without issue.
    
I have a total of 252 Leds for the segments and 14 leds for the dots. This makes a total of 266 Leds

The issue that I am facing is it only works when the Led length is set to 252 in the Led preference in wled settings, when changed to 266 it doesn't work after a reboot. If I set the length to 945 it doesn't work either.

Once we reboot wled, there is no boot animation anymore and the time nevers shows.

I think the problem could be a byte overflow when the led length is more than 255.

I'm not sure where to start looking to fix this issue, could you put me in the right direction?

### To Reproduce Bug

I have changed the code in the file const_ledclock.h as follow:

#define LC_LEDS_PER_SEGM 9 // LEDs per segment

#define LC_LEDMAP \
   -1, 206, 205, 204, 203, 202, 201, 200, 199, 198,  -1,       -1, 143, 142, 141, 140, 139, 138, 137, 136, 135,  -1,      -1,      -1,  80,  79,  78,  77,  76,  75,  74,  73,  72,  -1,    -1, 17, 16, 15, 14, 13, 12, 11, 10,  9, -1, \
  207,  -1,  -1,  -1,  -1,  -1,  -1,  -1,  -1,  -1, 197,      144,  -1,  -1,  -1,  -1,  -1,  -1,  -1,  -1,  -1, 134,     252,      81,  -1,  -1,  -1,  -1,  -1,  -1,  -1,  -1,  -1,  71,    18, -1, -1, -1, -1, -1, -1, -1, -1, -1,  8, \
  208,  -1,  -1,  -1,  -1,  -1,  -1,  -1,  -1,  -1, 196,      145,  -1,  -1,  -1,  -1,  -1,  -1,  -1,  -1,  -1, 133,     253,      82,  -1,  -1,  -1,  -1,  -1,  -1,  -1,  -1,  -1,  70,    19, -1, -1, -1, -1, -1, -1, -1, -1, -1,  7, \
  209,  -1,  -1,  -1,  -1,  -1,  -1,  -1,  -1,  -1, 195,      146,  -1,  -1,  -1,  -1,  -1,  -1,  -1,  -1,  -1, 132,     254,      83,  -1,  -1,  -1,  -1,  -1,  -1,  -1,  -1,  -1,  69,    20, -1, -1, -1, -1, -1, -1, -1, -1, -1,  6, \
  210,  -1,  -1,  -1,  -1,  -1,  -1,  -1,  -1,  -1, 194,      147,  -1,  -1,  -1,  -1,  -1,  -1,  -1,  -1,  -1, 131,     255,      84,  -1,  -1,  -1,  -1,  -1,  -1,  -1,  -1,  -1,  68,    21, -1, -1, -1, -1, -1, -1, -1, -1, -1,  5, \
  211,  -1,  -1,  -1,  -1,  -1,  -1,  -1,  -1,  -1, 193,      148,  -1,  -1,  -1,  -1,  -1,  -1,  -1,  -1,  -1, 130,     256,      85,  -1,  -1,  -1,  -1,  -1,  -1,  -1,  -1,  -1,  67,    22, -1, -1, -1, -1, -1, -1, -1, -1, -1,  4, \
  212,  -1,  -1,  -1,  -1,  -1,  -1,  -1,  -1,  -1, 192,      149,  -1,  -1,  -1,  -1,  -1,  -1,  -1,  -1,  -1, 129,     257,      86,  -1,  -1,  -1,  -1,  -1,  -1,  -1,  -1,  -1,  66,    23, -1, -1, -1, -1, -1, -1, -1, -1, -1,  3, \
  213,  -1,  -1,  -1,  -1,  -1,  -1,  -1,  -1,  -1, 191,      150,  -1,  -1,  -1,  -1,  -1,  -1,  -1,  -1,  -1, 128,     258,      87,  -1,  -1,  -1,  -1,  -1,  -1,  -1,  -1,  -1,  65,    24, -1, -1, -1, -1, -1, -1, -1, -1, -1,  2, \
  214,  -1,  -1,  -1,  -1,  -1,  -1,  -1,  -1,  -1, 190,      151,  -1,  -1,  -1,  -1,  -1,  -1,  -1,  -1,  -1, 127,      -1,      88,  -1,  -1,  -1,  -1,  -1,  -1,  -1,  -1,  -1,  64,    25, -1, -1, -1, -1, -1, -1, -1, -1, -1,  1, \
  215,  -1,  -1,  -1,  -1,  -1,  -1,  -1,  -1,  -1, 189,      152,  -1,  -1,  -1,  -1,  -1,  -1,  -1,  -1,  -1, 126,      -1,      89,  -1,  -1,  -1,  -1,  -1,  -1,  -1,  -1,  -1,  63,    26, -1, -1, -1, -1, -1, -1, -1, -1, -1,  0, \
   -1, 216, 217, 218, 219, 220, 221, 222, 223, 224,  -1,       -1, 153, 154, 155, 156, 157, 158, 159, 160, 161,  -1,      -1,      -1,  90,  91,  92,  93,  94,  95,  96,  97,  98,  -1,    -1, 27, 28, 29, 30, 31, 32, 33, 34, 35, -1, \
  251,  -1,  -1,  -1,  -1,  -1,  -1,  -1,  -1,  -1, 225,      188,  -1,  -1,  -1,  -1,  -1,  -1,  -1,  -1,  -1, 162,     259,     125,  -1,  -1,  -1,  -1,  -1,  -1,  -1,  -1,  -1,  99,    62, -1, -1, -1, -1, -1, -1, -1, -1, -1, 36, \
  250,  -1,  -1,  -1,  -1,  -1,  -1,  -1,  -1,  -1, 226,      187,  -1,  -1,  -1,  -1,  -1,  -1,  -1,  -1,  -1, 163,     260,     124,  -1,  -1,  -1,  -1,  -1,  -1,  -1,  -1,  -1, 100,    61, -1, -1, -1, -1, -1, -1, -1, -1, -1, 37, \
  249,  -1,  -1,  -1,  -1,  -1,  -1,  -1,  -1,  -1, 227,      186,  -1,  -1,  -1,  -1,  -1,  -1,  -1,  -1,  -1, 164,     261,     123,  -1,  -1,  -1,  -1,  -1,  -1,  -1,  -1,  -1, 101,    60, -1, -1, -1, -1, -1, -1, -1, -1, -1, 38, \
  248,  -1,  -1,  -1,  -1,  -1,  -1,  -1,  -1,  -1, 228,      185,  -1,  -1,  -1,  -1,  -1,  -1,  -1,  -1,  -1, 165,     262,     122,  -1,  -1,  -1,  -1,  -1,  -1,  -1,  -1,  -1, 102,    59, -1, -1, -1, -1, -1, -1, -1, -1, -1, 39, \
  247,  -1,  -1,  -1,  -1,  -1,  -1,  -1,  -1,  -1, 229,      184,  -1,  -1,  -1,  -1,  -1,  -1,  -1,  -1,  -1, 166,     263,     121,  -1,  -1,  -1,  -1,  -1,  -1,  -1,  -1,  -1, 103,    58, -1, -1, -1, -1, -1, -1, -1, -1, -1, 40, \
  246,  -1,  -1,  -1,  -1,  -1,  -1,  -1,  -1,  -1, 230,      183,  -1,  -1,  -1,  -1,  -1,  -1,  -1,  -1,  -1, 167,     264,     120,  -1,  -1,  -1,  -1,  -1,  -1,  -1,  -1,  -1, 104,    57, -1, -1, -1, -1, -1, -1, -1, -1, -1, 41, \
  245,  -1,  -1,  -1,  -1,  -1,  -1,  -1,  -1,  -1, 231,      182,  -1,  -1,  -1,  -1,  -1,  -1,  -1,  -1,  -1, 168,     265,     119,  -1,  -1,  -1,  -1,  -1,  -1,  -1,  -1,  -1, 105,    56, -1, -1, -1, -1, -1, -1, -1, -1, -1, 42, \
  244,  -1,  -1,  -1,  -1,  -1,  -1,  -1,  -1,  -1, 232,      181,  -1,  -1,  -1,  -1,  -1,  -1,  -1,  -1,  -1, 169,      -1,     118,  -1,  -1,  -1,  -1,  -1,  -1,  -1,  -1,  -1, 106,    55, -1, -1, -1, -1, -1, -1, -1, -1, -1, 43, \
  243,  -1,  -1,  -1,  -1,  -1,  -1,  -1,  -1,  -1, 233,      180,  -1,  -1,  -1,  -1,  -1,  -1,  -1,  -1,  -1, 170,      -1,     117,  -1,  -1,  -1,  -1,  -1,  -1,  -1,  -1,  -1, 107,    54, -1, -1, -1, -1, -1, -1, -1, -1, -1, 44, \
   -1, 242, 241, 240, 239, 238, 237, 236, 235, 234,  -1,       -1, 179, 178, 177, 176, 175, 174, 173, 172, 171,  -1,      -1,      -1, 116, 115, 114, 113, 112, 111, 110, 109, 108,  -1,    -1, 53, 52, 51, 50, 49, 48, 47, 46,  45, -1 
   
   
#define LC_SEP_LEDS 14
   
#define LC_SEP_LED_ROWS 1, 2,3,4,5,6,7,11,12,13,14,15,16,17
   
I also changed the DEFAULT_LED_COUNT in platformio_override.ini:
 -D DEFAULT_LED_COUNT=945
    
Here is an image of the led mapping for better visualisation:
[image]
    

### Expected Behavior

The Matrix should be working with 266 leds (matrix size of 945 leds)

### Install Method

Self-Compiled

### What version of WLED?

LedClock version 1.3.0 WLED version 0.14.1

### Which microcontroller/board are you seeing the problem on?

ESP32

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

imeszaros, 2024-02-12:
Hi! That must be a huge clock! :D Your assumption about the overflow is probably right. When I design this project I only considered a few leds per segment.

As per first tought, you should probably look for any `uint8_t` types within this files:

```
wled00\um_ledclock.h
wled00\7segmdisp.cpp
wled00\7segmdisp.h
```

and replace them with `uint16_t` maybe.

CommittotheIndian, 2024-02-12:
@Zaraxxx 

For what it's worth, I built that same clock and used this LED clock project but don't remember running into that LED count.

Zaraxxx, 2024-02-13:
Thanks a lot @imeszaros !
That fixed the issue. 
I would have make a fix branch to your repo but I have no permissions for that.

imeszaros, 2024-02-13:
Cool, feel free to fork the repo and make a pull request!

Zaraxxx, 2024-02-13:
Ignore my last comment, I'll redo it again since I originally forked wled.

Zaraxxx, 2024-02-17:
Pull Request created for you