imeszaros/ledclock #18: Visual Studio Code Issue

TMassing85, 2024-02-17:
I have an error in Visual Studio Code with PlatformIO and C++.  I don't know what I did wrong. Can you tell me how? I'm not good with code.

[image]

imeszaros, 2024-02-17:
Hi, please select the `ledclock` environment:

[image]

TMassing85, 2024-02-17:
Same problem both 1.1.0 and 1.3.0.

[image]

[image]

imeszaros, 2024-02-17:
[image]

See? This is your problem. I cannot tell you any other way. Change it to "ledclock" end it will compile.

closed 2024-02-17 by imeszaros